serialport/serialport-rs #270: Terminal settings not applied on OSX

emgardner, 2025-05-19:
cross linking an issue that I originally thought was related to `tokio_serial`: https://github.com/berkowski/tokio-serial/issues/83

The gist of it i was running into an issue where the same program was working fine on windows and linux for some reason on my M4 mac it isn't running after doing some troubleshooting and comparing serial port settings I was able to find a small difference. Turned out that just initializing  termios with the same settings fixed the problem.

```rust
// Original fix to replicate pyserial settings:
use anyhow::Result;
use std::time::Duration;
use tokio::io::{AsyncReadExt, AsyncWriteExt};
use tokio_serial::{DataBits, FlowControl, Parity, SerialPort, SerialPortBuilderExt, StopBits};

use nix::sys::termios::{self, InputFlags, SetArg, SpecialCharacterIndices};
use std::os::unix::io::{AsRawFd, BorrowedFd};

#[tokio::main]
async fn main() -> Result<()> {
   ...
    {
        let raw_fd = port.as_raw_fd();
        let borrowed: BorrowedFd<'_> = unsafe { BorrowedFd::borrow_raw(raw_fd) };
        let mut tio = termios::tcgetattr(&borrowed)?;
        tio.input_flags &= !(InputFlags::IGNBRK | InputFlags::IGNPAR);
        tio.control_chars[SpecialCharacterIndices::VMIN as usize] = 0;
        tio.control_chars[SpecialCharacterIndices::VTIME as usize] = 0;
        termios::tcsetattr(&borrowed, SetArg::TCSANOW, &tio)?;
    }
}
```

Then i tried to pare it down and see what actually mattered turned out this just worked:
```rust
    {
        let raw_fd = port.as_raw_fd();
        let borrowed: BorrowedFd<'_> = unsafe { BorrowedFd::borrow_raw(raw_fd) };
        let mut tio = termios::tcgetattr(&borrowed)?;
        termios::tcsetattr(&borrowed, SetArg::TCSANOW, &tio)?;
    }
```

```Cargo.toml
[package]
name = "client-testing"
version = "0.1.0"
edition = "2024"

[dependencies]
anyhow = "1.0.98"
tokio = { version = "1.45.0", features = ["full"] }
tokio-serial = { version = "5.4.1", features = ["bytes", "codec", "rt"] }
```

sirhcel, 2025-05-21:
Sorry to hear that @emgardner! Reading through this issue here and https://github.com/berkowski/tokio-serial/issues/83 I'm wondering if this might be related to #117. Do you know if this happens on other Apple hardware than the M4 as well? Or did you just test on Apple M4 hardware so far?

Could you please provide a self-contained example for reproducing it? For example as a Cargo project in its own Git repo? In case this affects other Apple hardware as well, then I could give it a deeper look from my side.

sirhcel, 2025-05-21:
Could you please give an example of what data you are trying to transmit and what you are actually receiving/gets on the wire @emgardner?

emgardner, 2025-05-22:
Hi @sirhcel here is the sample code i was using: https://github.com/emgardner/serialport-rs-repro a couple of things to note:
- I am using a PL2303 chip that interfaces with an IRDA transceiver 
- This only occurred after a dependency update where mio-serial and serialport-rs were bumped
- I've only tested this on the M4 Mac (Linux/Windows were Fine)
- I was receiving junk over the line values were oscillating between 255 and 248 when i was sending the packet in the example. I didn't pull out my logic analyzer to check the wire as i knew it had to be related to the software. 

Let me know if there's any additional help i can provide, would love to help get this resolved.

emgardner, 2025-05-23:
@sirhcel I can confirm this also fails on an M3 Max chip as well