serialport/serialport-rs #279: In some cases displays only fraction of real Serial Number of USB-Serial device

Spyglys3, 2025-07-31:
If in connected USB-Serial device real Serial number is formatted as "B4:3D:4D:B1:5A:C8" (as are in some ESP32-S3 Arduino boards), the plugin recognize only as "B4" and displays - serial_number: "B4".
Thus different boards with different real Serial numbers recognizes as the SAME board (same pid, same vid, and the same first fraction of serial_number).
At least on Windows 10 x64 platform.

sirhcel, 2025-08-02:
Thank you for spotting this @Spyglys3! We are parsing serial numbers from Windows opaque HWID strings and it looks like you stumbled upon a new case we were not aware of beforehand. We are learning as we go here. 😅

Could you please provide the full HWID string for your device in question so that we can come up with a test case and a fix? This information should be available in the device's details in the device manager.

Spyglys3, 2025-08-02:
Windows device manager shows:,
USB\VID_303A&PID_1001&MI_00
USB\VID_303A&PID_1001&REV_0101&MI_00

However, USB Device Tree Viewer V4.4.4 shows:
Device ID                : USB\VID_303A&PID_1001&MI_00\9&398007B8&0&0000
Hardware IDs          : USB\VID_303A&PID_1001&REV_0101&MI_00 USB\VID_303A&PID_1001&MI_00

Full information about the device from USB Device Tree Viewer V4.4.4 is attached. I hope it helps.
[USB info.txt](https://github.com/user-attachments/files/21558165/USB.info.txt)

sirhcel, 2025-08-15:
Thank you very much for collecting the data @Spyglys3! The attached data contains the device id which will be used for parsing and a fix is in the making with #282. Could you please give the charges from this PR a spin to check whether this actually solves your issue?

Spyglys3, 2025-08-15:
As I understand, in Cargo.toml I have change (I am very new in Rust):
```
[dependencies]
#serialport = {version = "4.7.0"} - original, not worked version 
serialport = {git = "https://github.com/sirhcel/serialport-rs.git"}
```
However, it does not work as well - the problem remains.

sirhcel, 2025-08-16:
You are almost there @Spyglys3. I'm assuming that omitting a revision makes Cargo pick the latest master branch.

Explicitly specifying the revision like
 
```
[dependencies]
serialport = { git = "https://github.com/sirhcel/serialport-rs.git", rev = "ff605ff1877405807b447a9d1c0dda9a2c47b7be" }
```

should pull in you the changes you are looking for. Could you please give it another spin?

Spyglys3, 2025-08-16:
Perfect. It works!
Thanks.

sirhcel, 2025-08-25:
Released with [4.7.3](https://github.com/serialport/serialport-rs/releases/tag/v4.7.3)

closed 2025-08-25 by sirhcel